esp-rs/esp32-hal #1: Initialize the repo

jeandudey, 2019-12-05:
At least an empty README.md, i want to give it a try by creating a simple HAL for GPIO pins

MabezDev, 2019-12-06:
Done :). I look forward to seeing what you come up wit.

closed 2019-12-06 by MabezDev

HarkonenBade, 2019-12-06:
Are you planning something based on https://github.com/rust-embedded/embedded-hal?

jeandudey, 2019-12-06:
@HarkonenBade yes, I'm planning to use the `embedded-hal` traits

HarkonenBade, 2019-12-06:
Awesome! Very interested to see more.

jeandudey, 2019-12-15:
I have been working today on creating a little `.svd` file by hand for the peripherals, mainly I'm going first to support GPIO, so svd2rust can be used. I'm not very optimistic since it's a lot of hard work, but as peripherals are in need, we can modify the `.svd` file adding the necessary peripherals and registers.

I've tried to mimic peripheral API by hand directly but it seems like an overkill and error prone.

And now that _Espressif_ announced on the `llvm-xtensa` repo that they're close to release a new version of the back-end I think that having a functional example is good showing the capabilities of Rust, and possibly get some new hands here.

HarkonenBade, 2019-12-15:
Would you consider putting what you have so far in a WIP pull request? As I have some time tomorrow and would be happy to add some registers.

jeandudey, 2019-12-15:
Let me finish some registers and I'll add a WIP PR.