esp-rs/esp32-hal #3: Using bindgen for an interface to the WiFi blobs

MabezDev, 2019-12-22:
The static libs found in `$IDF/components/esp32/lib` link against the headers inside  `$IDF/components/esp32/include`, using bindgen we should be able to wrap the c functions in a Rusty way.

HarkonenBade, 2019-12-23:
I started poking at this a bit, we might have issues interfacing with them as at least the wifi one seems to use some sort of async task structure? As I'm not sure if we can easily ensure that the required baggage for that is around in a rust program.

MabezDev, 2019-12-24:
Hmm it will be quite awkward, but just about doable I think. One of the idf developers replied to me about the wifi libs [on reddit](https://www.reddit.com/r/rust/comments/d23yl1/rust_on_the_esp32/ezvrh7z/?context=3) that explains a little bit more.

I suspect we may need more info, hopefully espressif don't mind helping on that front.

MabezDev, 2019-12-24:
Essentially we need to reimplement this file https://github.com/espressif/esp-idf/blob/master/components/esp32/esp_adapter.c , but it's not quite that straight forward as implementing requires:

- [ ] NVS driver
- [ ] Allocation, not sure how that will work with rust's `#[global_allocator]` or would we need to roll our own.
- [ ] Queue creation should be pretty simple, but there needs to a conversion wrapper that converts `* void` to The actual type that we can store on the rust side, I'm hoping only one type gets queued else this might get tricky.
- [ ] ISR management, Required work in `xtensa-lx6-rt` I think.
- [ ] For mutex's/ semaphores we should be able to get away with the spin lock crate.
- [ ]  Timers, we should be able to use model our code on the ets_timer functions in the idf that use the hardware timers
- [ ] Power & coex functions, should be reasonably simple to implement

arjanmels, 2020-07-17:
I have made a start at: https://github.com/arjanmels/esp32-wifi.
Bindgen is up and running and I can call some basic functions. Will start on the OS wrapper for the wifi_init function next.

arjanmels, 2020-09-11:
I have transferred my esp32_wifi repo to esp-rs: https://github.com/esp-rs/esp32-wifi.
Status: It allows todo wifi scan, but currently still returns 0 AP found. (Not certain what the problem is, I think all essential compatibility functions are in place, might be teh PHY not enabled properly.)

MabezDev, 2020-09-12:
Oh excellent! I started something similar, but will now put my efforts into esp-rs/esp32-wifi!