esp-rs/esp32-hal #17: uart0 tx fifo gets broken on startup

sapir, 2020-04-22:
I'm running code similar to the examples, and the TX FIFO gets out of sync: The right number of characters are written each loop iteration, but the current stream position is incorrect.

I think that what's happening here is that the bootloader writes some data into the FIFO, and then the `reset()` call in serial.rs reset the peripheral but doesn't entirely reset the FIFO. When the ESP-IDF driver initializes a uart, it doesn't reset it if it's uart0, and I think that maybe this is the reason.

I've found a few workarounds so far:

1. Commenting out the `reset()` call in serial.rs.
2. Resetting the TX FIFO by setting TX_FIFO_RST. This is only safe for UART0, because for UART1 it also clears UART2, and I think for UART2 it doesn't actually do anything. And also it probably messes up the output.
3. Waiting for the fifo to empty out. This messed up the previous output a bit. Possibly I didn't do it correctly, though.

arjanmels, 2020-04-23:
Thanks for your troubleshooting, I'll take a look at it this weekend.

arjanmels, 2020-06-22:
I cannot reproduce this (on current builds). Is there a particular exampel where you still notice this?

closed 2021-05-15 by MabezDev