esp-rs/esp32-hal #28: Rust bootloader

MabezDev, 2020-06-06:
We currently use the C bootloader used in the arduino sdk core (slightly modified for PSRAM support). We should be able to write an equivalent in Rust.

arjanmels, 2020-06-07:
Agree, however to add some color: it is quite a major task: the bootloader initializes flash, ram (depending on board and efuse config), memory protection unit; then it loads the partition table and loads the various segments to the correct places. Finally it sets up the caches correctly and hands over to teh user program.

MabezDev, 2020-06-07:
I agree this is a lot more complicated than it sounds. I'd say this is more a stretch goal, than something we should actively working on now.

MabezDev, 2021-12-23:
We can also think about using mcuboot instead of the IDF bootloader.