esp-rs/esp32-hal #53: ESP32-C3 support

mchodzikiewicz, 2021-02-10:
Hello there!

Have you heard about `ESP32-C3` which is `RISC-V` based?
I guess it removes major obstacle for ESP32 Rust adoption which is lack of upstream LLVM support for `xtensa-lx`. Do you plan any work on support for this chip anytime soon?

Assuming that peripherals are more or less the same I guess it would be fairly easy to make esp32-hal and esp32-wifi support the new chip.

Do you expect any major obstacles to run it on the new chip? (wifi stack binary blob bindings, proper linker magic to jump from the bootloader and so on)

pepijndevos, 2021-02-20:
I'm also curious about this. I have access to the evaluation hardware and have done a HAL before, but am completely unfamiliar with the ESP32 register definitions.

Are they mostly compatible, or does it make more sense to write a completely separate HAL for the RISC-V version?

I found there is already a PAC: https://github.com/imheresamir/esp32c3 so in an ideal world where the registers are completely backwards compatible, it would probably be minimal work to port it?

MabezDev, 2021-02-23:
> Do you expect any major obstacles to run it on the new chip? (wifi stack binary blob bindings, proper linker magic to jump from the bootloader and so on)

Hard to say at the moment, as I do not have access to a C3 dev kit, nor can I find a reference manual for the C3. I am guessing it's not been released yet.

pepijndevos, 2021-02-23:
Looking at the PAC documentation for esp32 and esp32c3 it seems they are far from compatible.
Compare:
https://docs.rs/esp32/0.10.0/esp32/gpio/index.html
https://docs.rs/esp32c3/0.1.2/esp32c3/gpio/index.html

The leaflet that came with the dev hardware confirms that documentation has not yet been released, and instructs to use the master branch of esp-idf together with the documentation of es32-s2 for setup.

It seems the PAC was generated from the headers in https://github.com/espressif/esp-idf/tree/master/components/esp32c3 but I'm not sure how much info is there to do the HAL.

MabezDev, 2021-02-23:
> Looking at the PAC documentation for esp32 and esp32c3 it seems they are far from compatible.
Compare:
https://docs.rs/esp32/0.10.0/esp32/gpio/index.html
https://docs.rs/esp32c3/0.1.2/esp32c3/gpio/index.html

This comparison is quite misleading, as the esp32 crate has had a [number of patches applied](https://github.com/esp-rs/esp32/tree/master/svd/patches) to clean up the output.

Looking a bit further and taking GPIO as an example, the `RegisterBlock` structs svd2rust generates are very compatible. As I suspected, Espressif have developed the C3 peripherals to be backwards compatible, only introducing new registers in 'reserved' space in the normal esp32. See [esp32 gpio](https://docs.rs/esp32/0.10.0/src/esp32/gpio.rs.html#1-233) & [esp32c3 gpio](https://docs.rs/esp32c3/0.1.2/src/esp32c3/gpio.rs.html#1-2577).

Of course I cannot verify this until I have hardware/ official docs from Espressif.

pepijndevos, 2021-02-26:
Oh interesting! Let's ping @imheresamir to get their input. Maybe the same patches can be applied to esp32c3?
I'd be happy to do some testing and fixes with my esp32c3.

FYI https://twitter.com/EspressifSystem was offering free developer hardware, so I'm sure they'd be happy to send you one.

imheresamir, 2021-02-26:
Hi, I haven't looked at those patches but it would be great if they were adaptable for the C3. I only got to the first step of getting an initial SVD so far.

I'd be happy to help out with the cleanup work on this pac, will try to find some time over the next few days.

t0bro, 2021-07-04:
Hi there!

Are there any good news about ESP32-C3 hal crate progress?

Thanks.

mchodzikiewicz, 2021-09-21:
Hi, I have just seen @MabezDev blog post about hello world on a C3 chip and just received a devkit.

I'm going to run https://github.com/MabezDev/esp32c3-experiments with intention of creating a branch of `esp32-hal` with `esp32c3` feature.

Any hints?

jessebraham, 2021-09-21:
@mchodzikiewicz there has been a little bit of discussion on this topic, this comment should hopefully summarize what we've been thinking:
https://github.com/esp-rs/esp-rs-backlog/issues/6#issuecomment-914639690

mchodzikiewicz, 2021-09-21:
> We now have an PAC for the esp32c3: https://github.com/esp-rs/esp32c3 btw :)
yep, I am going to use it :)

@jessebraham thanks, I have posted my view on it.

Playing around I have found a small issue already - I wanted to add `esp32` and `esp32c3` as crate features similar to STM32's convention in their HAL but dependencies and features cannot have the same names :( for now I have called them `device-esp32` and `device-esp32c3` but there might be a better solution for this