esp-rs/esp32-hal #56: UART carriage not returned

theLogicJB, 2021-03-18:
When printing multiple lines like this
```rust
    writeln!(serial, "ESP32 Started\n\n").unwrap();

    dprintln!("UART0");

    loop {
        writeln!(serial, "My Test line").unwrap();
        sleep(500.ms());
    }
```
I get this output
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0048,len:12
ho 0 tail 12 room 4
load:0x3fff0054,len:4800
load:0x40078000,len:17448
load:0x4007c428,len:4840
entry 0x4007c6a0
ESP32 Started

             UART0
                  My Test line
                              My Test line
                                          My Test line
```
Ending each string with `\r` solves the problem but shouldn't it be part of the library or am I missing something?

Is this related to #17?

chriss0612, 2021-03-18:
Thats just a problem with your terminal emulater interpreting line endings wrong / different and not a problem with the code itself

theLogicJB, 2021-03-20:
@chriss0612 Which terminal emulator do you know works? I tried Minicom or dterm inside Alacritty.

chriss0612, 2021-03-20:
picocom -b <baud> --imap lfcr /dev/ttyUSB0

should work i think, if not you can try with the --imap argument.

theLogicJB, 2021-03-23:
Even with `picocom` you need to remap line feed to carriage return with `--imap lfcr`. This means it is the third terminal emulator I try that needs special configuration to print the serial output in a readable way.
Wouldn't it be more user friendly to change it in the code instead of every user figuring it out on their own?

MabezDev, 2021-03-24:
@theLogicJB I'm not sure what you mean by change it in the code, the serial peripheral simply writes the bytes to tell it to out of the port. Like @chriss0612 mentioned, this is a display issue, I myself use pythons mterm script inside of alacritty and that works great. @kelnos has also just released [esp-monitor](https://github.com/kelnos/espmonitor) which may be of use to you.

closed 2021-03-24 by MabezDev