esp-rs/esp32-hal #57: Cannot import "i2c" from remote dependency, possible if local build

litchipi, 2021-06-16:
I got a code with this import line:
`use esp_hal::i2c::{self, Error, I2C}`

In the Cargo.toml file:
`esp32-hal = { path = "./esp32-hal/" }` (github clone in local dir) -> works fine
`esp32-hal = "0.2.0"` -> "could not find `i2c` in `esp32_hal`"

You can grab my full code here: https://github.com/litchipi/esp32rs/blob/oled_simple/src/oled_simple.rs
It's basically a copy-paste of the example given in the hal for now.

_(PS: I tried to create my own example, but the libs used in the example of the hal are outdated and the new versions doesn't work anymore. Should I open another issue for this as well ?)_

Dreistein, 2021-06-28:
The I2C feature is in the github repository but not on crates.io. I am not a maintainer but I believe the feature is not mature enough to be released. There is at least one todo in the i2c.rs file.

You could use the github repo in your ```Cargo.toml``` until the feature is on crates.io:
```esp32-hal = { git = "https://github.com/esp-rs/esp32-hal.git" }```

MabezDev, 2021-08-14:
The `0.3.0` now includes the I2C module :)

closed 2021-08-14 by MabezDev