esp-rs/esp32-hal #60: Baremetal 1.0.0

KerryRJ, 2021-08-06:
I have updated the baremetal crate to 1.0.0. This requires changes to svd2rust, xtensa-lx, esp32 and esp32-hal as the InterruptNumber trait was removed from baremetal. I have successfully tested the changes with the quickstart.

I have added the trait to xtensa-lx
```Rust
pub unsafe trait InterruptNumber: Copy {
    /// Return the interrupt number associated with this variant.
    ///
    /// See trait documentation for safety requirements.
    fn number(self) -> u16;
}
```
Is updating the baremetal crate acceptable? If it is, should this stay with the current fn nr(self) -> u16 or if fn number(self) -> u16 is acceptable?

MabezDev, 2021-08-06:
I think so - we will want to release a new `xtensa-lx{-rt}` etc release anyway. I welcome the PR on `xtensa-lx`.

KerryRJ, 2021-08-06:
I just went to upstream the svd2rust changes and noticed xtensa-lx 0.3.0 xtensa-lx-rt 0.5.0 are configured. Should I PR svd2rust changes with updated version numbers for those crates and if I do what should their version numbers be, or should I be PR'ing the xtensa-lx/-rt updates before svd2rust? If I PR lx/-rt first we bumping the minor or release numbers, i.e. 0.3.1 and 0.5.1?

MabezDev, 2021-08-06:
Yes, we'll want to work our way up from the bottom of the chain, so `xtensa-lx` & `xtensa-lx-rt` first. I can make a release once I merge https://github.com/esp-rs/xtensa-lx-rt/pull/23.

MabezDev, 2021-08-06:
In my zombie like state I just published `0.6.0` of xtensa-lx-rt without your PR and without bumping bare metal :facepalm:. Would you mind sending your PR over so I can release `0.7.0`

KerryRJ, 2021-08-07:
Sent the PR.

MabezDev, 2021-08-13:
Release in 0.3.0

closed 2021-08-13 by MabezDev