esp-rs/esp32-hal #63: Example ram RTCWDT_RTC_RESET

KerryRJ, 2021-08-13:
I pulled down the latest crates and successfully ran alloc and mem(thank you) but ram is now failing

> ets Jun  8 2016 00:22:57

>rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0048,len:12
ho 0 tail 12 room 4
load:0x3fff0054,len:4800
load:0x40078000,len:17448
load:0x4007c428,len:4840
entry 0x4007c6a0
ets Jun  8 2016 00:22:57

>rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0048,len:12
ho 0 tail 12 room 4
load:0x3fff0054,len:4800
load:0x40078000,len:17448
load:0x4007c428,len:4840
entry 0x4007c6a0

Are you able to confirm it working on your side @MabezDev?

KerryRJ, 2021-08-13:
Info from Jtag session

> Reading symbols from C:\Dev\Github\esp32-hal\target\xtensa-esp32-none-elf\debug\examples\ram...
done.
0x40000288 in ?? ()
Program stopped, probably due to a reset and/or halt issued by debugger
Register cache flushed.

After a debug restart

> JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
esp32.cpu0: Debug controller was reset.
esp32.cpu0: Core was reset.
esp32.cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
esp32.cpu0: Core was reset.
esp32.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
esp32.cpu1: Debug controller was reset.
esp32.cpu1: Core was reset.
esp32.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
Register cache flushed.
Continuing.
Note: automatically using hardware breakpoints for read-only addresses.
esp32.cpu0: Target halted, PC=0x400003C0, debug_reason=00000001
Set GDB target to 'esp32.cpu0'
esp32.cpu1: Target halted, PC=0x400076DD, debug_reason=00000000

>Program
 received signal SIGTRAP, Trace/breakpoint trap.
0x400003c0 in ?? ()

I cannot get a breakpoint at the start of main to work.

MabezDev, 2021-08-13:
Works for me:
`cargo +esp-dev espflash --tool cargo --chip esp32 --speed 460800 --example mem --features alloc /dev/ttyUSB0`
```rust
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0048,len:12
ho 0 tail 12 room 4
load:0x3fff0054,len:4800
load:0x40078000,len:17448
load:0x4007c428,len:4840
entry 0x4007c6a0
␀

Reboot!

Initializing
                 initialized src: 131072: 0.527s, 242.758KB/s
           memset aligned, sized 4 bytes: 0.307s, 8327.475KB/s
                          memset aligned: 0.308s, 8319.014KB/s
                                  memset: 0.308s, 8321.861KB/s
                      memcpy: 0 0 131072: 0.245s, 10433.002KB/s, Result: 0
                      memcpy: 0 0 131071: 0.245s, 10430.448KB/s, Result: 0
                      memcpy: 1 1 131071: 0.245s, 10428.311KB/s, Result: 0
                      memcpy: 1 0 131071: 0.267s, 9577.517KB/s, Result: 0
                      memcpy: 0 1 131071: 0.267s, 9578.145KB/s, Result: 0
                      memcpy: 0 0 131072: 0.234s, 10919.378KB/s, Result: 0
                      memcpy: 0 0 131071: 0.235s, 10910.983KB/s, Result: 0
                      memcpy: 1 1 131071: 0.235s, 10911.427KB/s, Result: 0
                      memcpy: 1 0 131071: 0.254s, 10089.640KB/s, Result: 0
                      memcpy: 0 1 131071: 0.254s, 10090.522KB/s, Result: 0
Alive and waiting for watchdog reset
Alive and waiting for watchdog reset
Alive and waiting for watchdog reset
Alive and waiting for watchdog reset
```

Are you enabling the `external-ram` feature? If you are, without psram connected that will almost certainly end in trouble :D

MabezDev, 2021-08-13:
Scratch what I said above, not working for me either!