esp-rs/esp32-hal #64: Linking error when enabling GPIO_INTR

schphil, 2021-09-22:
Hi there, attempting to enable the GPIO_INTR ( `esp32_hal::interrupt::enable(Interrupt::GPIO_INTR);` ) results in the following linking error. Thanks in advance.

[image]

MabezDev, 2021-09-22:
Are you trying to use esp32-hal in a std application, like https://github.com/ivmarkov/rust-esp32-std-hello? Whilst I'd like this to work in the future, this is currently not a supported configuration.

I would say use the [idf-hal](https://github.com/esp-rs/esp-idf-hal/blob/master/src/gpio.rs) implementation but this does not have support for Interrupts just yet.

schphil, 2021-09-22:
Thanks for the quick response. That is exactly what I am trying to do. Yes, I noticed that too which is why I tried the `esp32-hal` crate. I'll leave the interrupts for now. 

Should I close this one?

closed 2021-12-23 by MabezDev