esp-rs/esp32-hal #70: ESP32 doesn't boot when using external_ram with opt-level to 0, 1, or 2

sepotvin, 2021-12-11:
Hi,

When using the "external-ram" feature, the board (ESP32) will not boot (infinite watchdog loop) if the opt-level is set to 0, 1, or 2. Setting the opt-level to 3, "s", or "z" will fix the issue. Removing the "external-ram" feature will make the board boot at all optimization levels. The attached project is a minimal example to reproduce the problem. I'm using the rust compiler from esp-rs/rust on the esp-1.57.0.2 branch with the esp-idf on branch release/v4.3

[esp32_extram_error.zip](https://github.com/esp-rs/esp32-hal/files/7696073/esp32_extram_error.zip)

MabezDev, 2021-12-11:
Thanks for the report. Which dev board are you using with the esp32?

sepotvin, 2021-12-11:
I have tried on two different boards with the same behavior.

- An old TTGO T-Camera Plus with a 4MB SPRAM (instead of the 8 on the new boards). The ESP32 is a revision 1 chip.
- DOIT ESP32 DEVKITV1. The ESP32 is a revision 1 chip.

I also tried with the latest master branch of ESP-IDF (v5.0-dev-676-g5c33570524) without any changes.