esp-rs/esp32-hal #71: Blinky example not working

Raunak-Singh-Inventor, 2021-12-18:
I ran `cargo espflash --example blinky --release /dev/ttyUSB0`.

But the LED on Pin 13 is not blinking. Here is the output from `espmonitor`:
```
Opening /dev/ttyUSB0 with speed 115200
Using /home/raunaksingh/RustProjects/esp32-hal/target/xtensa-esp32-none-elf/release/examples/blinky as flash image
Resetting device... done
ets Jul 29 2019 12:21:46
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0048,len:12
ho 0 tail 12 room 4
load:0x3fff0054,len:4800
load:0x40078000 [_stack_end_cpu0:??:??],len:17448
load:0x4007c428 [_stack_end_cpu0:??:??],len:4840
entry 0x4007c6a0 [_stack_end_cpu0:??:??]
```

Any help would be appreciated.

jessebraham, 2021-12-18:
The `blinky` example is using GPIO2 instead of the more common GPIO13, as seen here:  
https://github.com/esp-rs/esp32-hal/blob/master/examples/blinky.rs#L31

Changing this line should hopefully resolve the issue.

Raunak-Singh-Inventor, 2021-12-18:
Thanks for the help. It worked.

Suggest that we include this instructions in the README.md on the main page. Should I create a pull request if its okay?

closed 2021-12-18 by Raunak-Singh-Inventor