esp-rs/esp32-hal #72: Enable the Ack for coded phy advertising

iamcalledusman, 2021-12-25:
I am using BLE 5.0 coded PHY advertising type is "ESP_BLE_GAP_SET_EXT_ADV_PROP_SCANNABLE". Able to send messages successfully and read via BLE50_Client example. My question is that, is it possible to enable the ACK? In case no ACK receives then retransmission of the same packet to the receiver.

Thanks In advance

MabezDev, 2022-01-04:
Could you clarify exactly what you're doing now, and what you need from this repo to achieve your goal?

iamcalledusman, 2022-01-04:
Thank you for the response!

let me explain it in more detail.

The requirement is to send the advertisement/beacon packets in the connectable mode using **Coded Phy** so that we can get **ACK** from the receiver to ensure the packet is received at the receiver end

Able to send packets using **coded PHY** using esp-idf examples (Periodic_adv/ Multi-adv as a Transmitter) & (Periodic_sync /BLE50_Security_Client as a Receiver).

The advertising type used for Coded PHY:   "**ESP_BLE_GAP_SET_EXT_ADV_PROP_SCANNABLE**".

This is non-connectable and scannable. But when we set advertising type as per our requirement which is **connectable** using Coded Phy 

"**ESP_BLE_GAP_SET_EXT_ADV_PROP_CONNECTABLE**" generates errors at run time with coded PHY. 
[image]
I have attached the image as well.

Can you please mention an example code or modification in the above-mentioned examples?

Thanks in Advance!

MabezDev, 2022-01-04:
Please file this issue under https://github.com/espressif/esp-idf. This Repo is for using Rust.

closed 2022-01-04 by MabezDev