esp-rs/esp32-hal #73: esp32 c3 coded phy double messages

iamcalledusman, 2021-12-25:
Hi All!

I am using multi-adv example ( esp-idf ) for esp32c3 . In which i am sending the coded phy messages and receiving from ble5-_security_client example.

Tx Side i am sending the following data using coded phy . " ESP_MULTI_ADV_CODED"

and on Rx side i am receiving

I (8315684) SEC_GATTC_DEMO: extend adv, adv type 0x2 data len rssi -51
I (8315694) SEC_GATTC_DEMO: extend adv, adv type 0xa data len
ESP_MULTI_ADV_CODED rssi -51

My question is that I should receive the single message which is

(8315694) SEC_GATTC_DEMO: extend adv, adv type 0xa data len
ESP_MULTI_ADV_CODED rssi -51

why I am receiving this packet as well.

I (8315684) SEC_GATTC_DEMO: extend adv, adv type 0x2

Thanks in Advance .

MabezDev, 2022-01-04:
Is this issue related to Rust?

iamcalledusman, 2022-01-04:
Thank you for the response! 

The issue is related to the esp32c3, embedded C. Sorry I don't know about Rust.

closed 2022-01-04 by jessebraham