esp-rs/esp32-hal #75: Gpio output is_set_high() not showing the correct value

alvaroarenas, 2022-01-02:
Hi there,

I am new to Rust and I started playing with my ESP32. I am trying to toggle a led connected to gpio2 by messaging between threads using `mpsc::channel`. I noticed that although I toggle the pin the `is_set_high()` always returns `false`.

I am using the code below. Which does toggle the led because it uses `counter` to keep the state. But I would like to use the state the gpio has, so that I can use multiple threads.  I tried initiating the led pin outside the thread and moving it with an Arc<Mutex<>> with the same result. What am I doing wrong ? or could it be that there is a bug in the library implementation?

```rust

    let (tx, rx) = mpsc::channel();

    thread::spawn(move || {
        let peripherals = target::Peripherals::take().expect("Failed to obtain Peripherals");
        let pins = peripherals.GPIO.split();
        let mut led = pins.gpio2.into_push_pull_output();
        let mut counter = 1;
        loop {
            let msg = rx.recv().unwrap();
            println!("Received message {}:{}", counter, msg);

            // let mut led = led.lock().unwrap();
            println!("led was high? {}", led.is_set_high().unwrap()); // always prints false
            // led.toggle();
            if counter % 2 == 0 {
                info!("toogle LOW");
                led.set_low();
            } else {
                info!("toogle HIGH");
                led.set_high();
            }
            println!("led IS high? {}", led.is_set_high().unwrap()); // always prints false although I just toggled it
            counter += 1;
        }
    });
```

I am using esp32-hal version 0.3.0.

alvaroarenas, 2022-01-02:
UPDATE: I think the bug is in the library. I modified the blinky example as below and the light stays constant

```
    loop {
        led.toggle().unwrap();
        delay(CORE_HZ);
        led.toggle().unwrap();
        delay(CORE_HZ);
    }
```

jessebraham, 2022-01-04:
I had recently fixed this same issue in `esp-idf-hal` as well, forgot to submit the changes here:
https://github.com/esp-rs/esp-idf-hal/pull/18/files

I will open a PR later today or this week addressing this.

PR #76 (2022-01-06): Fix `StatefulOutputPin` implementation, simplify `ToggleableOutputPin` implementation
Fixes #75.

Also replaced the implementation of the `ToggleableOutputPin` trait with `toggleable::Default`, as they're identical.
closed 2022-01-06 by MabezDev